JTA-STAR/J-STAR #18: package for Debian Desktop on Linux (virtual/real) PC?

jg1uaa, 2024-03-23:
Desktop Pi-Star looks good idea and I think it can be applied to Debian Desktop on x86 virtual machine such as VirtualBox (maybe VMware) or real machine, with MMDVM-USB HAT.
How do you think about this idea?

charugaa, 2024-03-26:
How to install J-STAR on a classic Debian x64 machine without VirtualBox.

bi7jta, 2024-03-27:
> Desktop Pi-Star looks good idea and I think it can be applied to Debian Desktop on x86 virtual machine such as VirtualBox (maybe VMware) or real machine, with MMDVM-USB HAT. How do you think about this idea?

Hi, 
download from https://github.com/JTA-STAR/J-STAR/blob/main/README.md 
then  , click "update" sync the new version.

VMware, ESXi need you export from VisualBox yourself , I think not a hard work.

bi7jta, 2024-03-27:
> Hi, download from https://github.com/JTA-STAR/J-STAR/blob/main/README.md then , click "update" sync the new version.
>
> VMware, ESXi need you export from VisualBox yourself , I think not a hard work.

> How to install J-STAR on a classic Debian x64 machine without VirtualBox.

> How to install J-STAR on a classic Debian x64 machine without VirtualBox.

Hi, 
download from https://github.com/JTA-STAR/J-STAR/blob/main/README.md 
then  , click "update" sync the new version.

No Debian x64 Physical machine plan, main purpose is Raspberry Pi 4B,5B ,2W. 
X86 VM is for J-STAR + FreeDMR server, RustDesk server.

closed 2024-03-27 by bi7jta