JTA-STAR/J-STAR #25: FM Network will Automatically trigger PTT , unknow reason

bi7jta, 2024-04-14:
**Describe the bug**
 FM Network will Automatically trigger PTT , when the 5V2V power supply is unstable

Listening in Allstarlink sip VoIP, will hear the noise , and will also will TX to DVSwitch Analog_bridge, MMDVM_bridge, to DMR network.

What I have made before this issue appear? I clone this J-STAR OS to another SD/TF card.

Possible reasons:

1,  the 5V2V power supply is unstable

2, SD card problem.

Video:  https://youtu.be/mbbfl4YhuI8

**logs**
` ``
tail -100f /var/log/pi-star/MMDVM-2024-04-14.log 
E: 2024-04-14 00:00:56.427 MMDVM ADC levels have overflowed
E: 2024-04-14 00:00:56.680 MMDVM ADC levels have overflowed
E: 2024-04-14 00:15:04.812 MMDVM ADC levels have overflowed
E: 2024-04-14 00:15:05.062 MMDVM ADC levels have overflowed
E: 2024-04-14 00:29:46.439 MMDVM ADC levels have overflowed
E: 2024-04-14 00:29:46.691 MMDVM ADC levels have overflowed
E: 2024-04-14 00:32:06.687 MMDVM ADC levels have overflowed
E: 2024-04-14 00:32:06.941 MMDVM ADC levels have overflowed
E: 2024-04-14 00:32:12.983 MMDVM ADC levels have overflowed
E: 2024-04-14 00:32:13.235 MMDVM ADC levels have overflowed
E: 2024-04-14 00:34:12.073 MMDVM ADC levels have overflowed
E: 2024-04-14 00:34:12.323 MMDVM ADC levels have overflowed
E: 2024-04-14 00:35:35.638 MMDVM ADC levels have overflowed
E: 2024-04-14 00:35:35.891 MMDVM ADC levels have overflowed
E: 2024-04-14 00:58:44.054 MMDVM ADC levels have overflowed
E: 2024-04-14 00:58:44.304 MMDVM ADC levels have overflowed
E: 2024-04-14 00:58:47.834 MMDVM ADC levels have overflowed
E: 2024-04-14 00:58:48.085 MMDVM ADC levels have overflowed
E: 2024-04-14 00:59:01.684 MMDVM ADC levels have overflowed
E: 2024-04-14 00:59:01.937 MMDVM ADC levels have overflowed
E: 2024-04-14 00:59:24.594 MMDVM ADC levels have overflowed
E: 2024-04-14 00:59:24.845 MMDVM ADC levels have overflowed
E: 2024-04-14 00:59:32.894 MMDVM ADC levels have overflowed
E: 2024-04-14 00:59:33.146 MMDVM ADC levels have overflowed
E: 2024-04-14 00:59:36.424 MMDVM ADC levels have overflowed
E: 2024-04-14 00:59:36.677 MMDVM ADC levels have overflowed
E: 2024-04-14 00:59:40.969 MMDVM ADC levels have overflowed
E: 2024-04-14 00:59:41.219 MMDVM ADC levels have overflowed
E: 2024-04-14 00:59:49.534 MMDVM ADC levels have overflowed
E: 2024-04-14 00:59:49.788 MMDVM ADC levels have overflowed
E: 2024-04-14 01:00:31.834 MMDVM ADC levels have overflowed
E: 2024-04-14 01:00:32.085 MMDVM ADC levels have overflowed
E: 2024-04-14 01:01:30.499 MMDVM ADC levels have overflowed
E: 2024-04-14 01:01:30.749 MMDVM ADC levels have overflowed
E: 2024-04-14 01:01:32.259 MMDVM ADC levels have overflowed
E: 2024-04-14 01:01:32.513 MMDVM ADC levels have overflowed
E: 2024-04-14 01:01:32.764 MMDVM ADC levels have overflowed
E: 2024-04-14 01:01:33.016 MMDVM ADC levels have overflowed
E: 2024-04-14 01:01:33.519 MMDVM ADC levels have overflowed
E: 2024-04-14 01:01:33.769 MMDVM ADC levels have overflowed
E: 2024-04-14 01:01:34.019 MMDVM ADC levels have overflowed
E: 2024-04-14 01:04:19.284 MMDVM ADC levels have overflowed
E: 2024-04-14 01:04:19.535 MMDVM ADC levels have overflowed
E: 2024-04-14 01:04:20.540 MMDVM ADC levels have overflowed
E: 2024-04-14 01:04:20.790 MMDVM ADC levels have overflowed
E: 2024-04-14 01:05:10.664 MMDVM ADC levels have overflowed
E: 2024-04-14 01:05:10.918 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:02.235 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:02.487 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:02.989 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:03.244 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:07.279 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:07.533 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:19.370 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:19.622 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:20.885 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:21.137 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:21.391 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:21.642 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:22.400 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:22.653 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:22.905 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:23.158 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:26.685 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:26.939 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:34.754 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:35.009 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:35.764 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:36.019 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:38.291 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:38.794 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:39.045 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:40.045 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:40.300 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:42.319 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:42.570 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:42.824 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:43.075 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:45.349 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:45.601 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:53.645 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:53.897 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:54.400 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:54.655 MMDVM ADC levels have overflowed
E: 2024-04-14 01:18:54.909 MMDVM ADC levels have overflowed
E: 2024-04-14 01:19:03.224 MMDVM ADC levels have overflowed
E: 2024-04-14 01:19:03.478 MMDVM ADC levels have overflowed
E: 2024-04-14 01:19:15.820 MMDVM ADC levels have overflowed
E: 2024-04-14 01:19:16.070 MMDVM ADC levels have overflowed
E: 2024-04-14 01:19:36.969 MMDVM ADC levels have overflowed
E: 2024-04-14 01:19:37.224 MMDVM ADC levels have overflowed

```
**Screenshots**
 
![image](https://github.com/JTA-STAR/J-STAR/assets/22002824/408d8587-24b4-4acf-9da9-e507bfd8c209)

MMDVM Bridge log

![image](https://github.com/JTA-STAR/J-STAR/assets/22002824/0be69709-dae0-49b7-ab7f-e454fa338ec4)

Analog Bridge log

![image](https://github.com/JTA-STAR/J-STAR/assets/22002824/9f72518b-9db4-4ca0-a9f7-0986d8e629e1)
 
USRP Allstarlink log:

bi7jta, 2024-04-14:
Excluding the "SD card clone " reason.

```
D: 2024-04-14 02:44:42.913 FM Network Data Sent
D: 2024-04-14 02:44:42.913 0000:  55 53 52 50 00 00 00 00 00 00 00 00 00 00 00 00    *USRP............*
D: 2024-04-14 02:44:42.913 0010:  00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00    *................*
D: 2024-04-14 02:44:42.914 0020:  08 13 00 00 00 00 00 00 00 00 00 00 00 00 42 49    *..............BI*
D: 2024-04-14 02:44:42.914 0030:  37 4A 54 41 00 00 00 00 00 00 00 00 00 00 00 00    *7JTA............*
D: 2024-04-14 02:44:42.914 0040:  00 00 00 00 00 00                                  *......*
D: 2024-04-14 02:44:42.914 FM Network Data Sent
D: 2024-04-14 02:44:42.914 0000:  55 53 52 50 00 00 00 00 00 00 00 00 00 00 00 01    *USRP............*
D: 2024-04-14 02:44:42.914 0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    *................*
D: 2024-04-14 02:44:42.914 0020:  17 F6 B6 D4 52 B2 EE AE B5 BD EC CF B7 01 D2 52    *....R..........R*
D: 2024-04-14 02:44:42.914 0030:  31 67 2A 2F 95 15 3E 19 37 0E 29 11 21 17 6E 14    *1g*/..>.7.).!.n.*
D: 2024-04-14 02:44:42.914 0040:  D0 18 CA 19 B6 16 17 16 68 12 C4 0D 39 1B 1A 3E    *........h...9..>*
D: 2024-04-14 02:44:42.914 0050:  EF 38 84 FE 6E DD 83 D5 25 C6 51 C5 A2 CA 43 CC    *.8..n...%.Q...C.*
D: 2024-04-14 02:44:42.914 0060:  4A EE 29 30 6D 36 FD F2 2B EA A2 22 D4 16 19 D2    *J.)0m6..+.."....*
D: 2024-04-14 02:44:42.914 0070:  C7 B7 3A B8 B6 B5 28 C4 FF D5 19 E4 D4 F7 20 20    *..:...(.......  *
D: 2024-04-14 02:44:42.914 0080:  A3 64 17 70 51 2C B7 09 F1 06 87 F5 F7 F5 26 FC    *.d.pQ,........&.*
D: 2024-04-14 02:44:42.914 0090:  A0 F9 E9 FE 5F 02 41 03 78 07 E6 08 7C 09 9F 0A    *...._.A.x...|...*
D: 2024-04-14 02:44:42.914 00A0:  BF 09 0A 21 C6 57 D6 54 84 07 CB DD B0 D5 AC C2    *...!.W.T........*
D: 2024-04-14 02:44:42.914 00B0:  12 C3 C9 CA AA CC 84 D8 64 E3 C6 EB 74 F7 1A 00    *........d...t...*
D: 2024-04-14 02:44:42.914 00C0:  53 07 65 0E BB 12 2A 16 94 18 2E 19 02 31 3E 6A    *S.e...*......1>j*
D: 2024-04-14 02:44:42.914 00D0:  13 6E E2 1F 45 EB 3A DC 8D C5 04 C1 64 C9 69 D0    *.n..E.:.....d.i.*
D: 2024-04-14 02:44:42.914 00E0:  42 DB BA DE 95 E7 EF FC 82 04 2F FF EE FE 9C FE    *B........./.....*
D: 2024-04-14 02:44:42.914 00F0:  AF 00 7E 23 3A 62 8B 63 01 1B 5B EF DA DE 01 CC    *..~#:b.c..[.....*
D: 2024-04-14 02:44:42.914 0100:  EC D6 D9 FB 1C 35 36 40 04 FC 12 F0 BD 29 F2 1A    *.....56@.....)..*
D: 2024-04-14 02:44:42.914 0110:  36 D4 70 D1 FE 06 08 02 B8 C3 2C B4 18 CE 86 CF    *6.p.......,.....*
D: 2024-04-14 02:44:42.914 0120:  14 C8 4D D1 9E DF 13 EF FC 00 CA 10 4F 1E A7 29    *..M.........O..)*
D: 2024-04-14 02:44:42.914 0130:  22 31 C1 35 7C 37 12 36 A1 32 4F 2D 4F 26 2E 21    *"1.5|7.6.2O-O&.!*
D: 2024-04-14 02:44:42.914 0140:  2D 1F AD 16 ED 05 4B 10 FE 3C 85 34 B1 01 E9 11    *-.....K..<.4....*
D: 2024-04-14 02:44:42.914 0150:  66 09 93 AD 8E 88 D6 A4 85 D0 23 E1 8E AA C1 89    *f.........#.....*
D: 2024-04-14 02:44:42.933 FM Network Data Sent
D: 2024-04-14 02:44:42.933 0000:  55 53 52 50 00 00 00 01 00 00 00 00 00 00 00 01    *USRP............*
D: 2024-04-14 02:44:42.933 0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    *................*
D: 2024-04-14 02:44:42.934 0020:  8D 9F 14 B1 E8 E3 CB 3A 35 4E 68 1A 3C 16 27 42    *.......:5Nh.<.'B*
D: 2024-04-14 02:44:42.934 0030:  76 3B AF 0F 50 FF F3 FD 1E FA AD FF 49 06 F1 09    *v;..P.......I...*
D: 2024-04-14 02:44:42.934 0040:  B0 10 CD 15 CA 17 CB 17 40 12 8C 07 EA 04 41 09    *........@.....A.*
D: 2024-04-14 02:44:42.934 0050:  E4 02 F2 FF C8 0A AB 25 FB 59 9A 58 3A 08 AD F3    *.......%.Y.X:...*
D: 2024-04-14 02:44:42.934 0060:  F6 22 12 0C 5E BE 9F B5 BB E8 3E E2 4E A3 CA A5    *."..^.....>.N...*
D: 2024-04-14 02:44:42.934 0070:  03 E8 47 EC 35 B6 52 C1 42 08 C5 0F 54 DB 85 E5    *..G.5.R.B...T...*
D: 2024-04-14 02:44:42.934 0080:  9A 2A 05 2F 6F F6 FB FB 99 3B B6 3A 75 FD 33 E3    *.*./o....;.:u.3.*
D: 2024-04-14 02:44:42.934 0090:  F0 DF 18 DC CF EF A8 03 0D 0F DF 1D E0 22 0E 25    *.............".%*
D: 2024-04-14 02:44:42.934 00A0:  FC 29 C7 28 7F 26 30 24 94 1E B6 18 46 12 1A 0B    *.).(.&0$....F...*
D: 2024-04-14 02:44:42.934 00B0:  B7 04 56 FE 70 F8 BA F3 B5 EF A5 EC B2 EA A3 E9    *..V.p...........*
D: 2024-04-14 02:44:42.934 00C0:  6D E9 F2 E9 40 EB 4E ED C6 EF 75 F2 56 F5 5D F8    *m...@.N...u.V.].*
D: 2024-04-14 02:44:42.934 00D0:  46 FB CC FD 09 00 93 0C 5B 26 40 27 C4 07 61 00    *F.......[&@'..a.*
D: 2024-04-14 02:44:42.934 00E0:  B9 12 97 0A EB EE 1D E7 8E ED 93 EB 74 E5 A7 FD    *............t...*
D: 2024-04-14 02:44:42.934 00F0:  81 38 61 3C 0E F9 FE D5 70 CF BF C3 28 D2 87 E0    *.8a<....p...(...*
D: 2024-04-14 02:44:42.934 0100:  6E E1 19 F1 7C 03 37 18 4F 39 6C 3F D6 23 85 28    *n...|.7.O9l?.#.(*
D: 2024-04-14 02:44:42.934 0110:  3C 50 EE 45 2C 09 2E 01 6C 2F 07 21 62 D7 A8 B5    *<P.E,...l/.!b...*
D: 2024-04-14 02:44:42.934 0120:  00 B2 95 A9 4A B3 1C DB 81 1E 4E 30 10 F5 E2 F2    *....J.....N0....*
D: 2024-04-14 02:44:42.934 0130:  1D 33 EC 2A EC E8 BB CE 69 CA A8 C6 8E D8 61 EA    *.3.*....i.....a.*
D: 2024-04-14 02:44:42.934 0140:  C3 F5 1A 07 5A 13 46 19 BE 1B BA 1F 95 2A 36 39    *....Z.F......*69*
D: 2024-04-14 02:44:42.934 0150:  95 4C AD 4A FB 28 DE 12 7A 0A A1 FB 65 F3 21 EF    *.L.J.(..z...e.!.*
D: 2024-04-14 02:44:42.953 FM Network Data Sent
D: 2024-04-14 02:44:42.953 0000:  55 53 52 50 00 00 00 02 00 00 00 00 00 00 00 01    *USRP............*
[... cut ...]
D: 2024-04-14 02:44:43.900 0110:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    *................*
D: 2024-04-14 02:44:43.900 0120:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    *................*
D: 2024-04-14 02:44:43.900 0130:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    *................*
D: 2024-04-14 02:44:43.900 0140:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    *................*
D: 2024-04-14 02:44:43.900 0150:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    *................*
E: 2024-04-14 02:44:44.121 MMDVM ADC levels have overflowed

```

bi7jta, 2024-04-14:
Disable FM Network, you can see the "FM Mode Automatically be trigger" , but no PTT to RF, seem the sound card settings problem.

```
I: 2024-04-14 03:24:12.126 MMDVMHost-20230212 is running

E: 2024-04-14 03:27:53.140 MMDVM ADC levels have overflowed
E: 2024-04-14 03:27:53.391 MMDVM ADC levels have overflowed
E: 2024-04-14 03:28:05.475 MMDVM ADC levels have overflowed
E: 2024-04-14 03:28:05.727 MMDVM ADC levels have overflowed
E: 2024-04-14 03:29:19.476 MMDVM ADC levels have overflowed
E: 2024-04-14 03:29:19.728 MMDVM ADC levels have overflowed
E: 2024-04-14 03:29:52.720 MMDVM ADC levels have overflowed
E: 2024-04-14 03:29:52.972 MMDVM ADC levels have overflowed
E: 2024-04-14 03:30:18.391 MMDVM ADC levels have overflowed
E: 2024-04-14 03:30:18.644 MMDVM ADC levels have overflowed
E: 2024-04-14 03:32:10.913 MMDVM ADC levels have overflowed
E: 2024-04-14 03:32:11.166 MMDVM ADC levels have overflowed
E: 2024-04-14 03:32:13.686 MMDVM ADC levels have overflowed
E: 2024-04-14 03:32:13.936 MMDVM ADC levels have overflowed
E: 2024-04-14 03:32:58.231 MMDVM ADC levels have overflowed
E: 2024-04-14 03:32:58.483 MMDVM ADC levels have overflowed
```

```
[FM]
Enable=1
Callsign=BI7JTA
CallsignSpeed=20
CallsignFrequency=1000
CallsignTime=10
CallsignHoldoff=0
CallsignHighLevel=50
CallsignLowLevel=20
CallsignAtStart=1
CallsignAtEnd=1
CallsignAtLatch=0
RFAck=K
ExtAck=N
AckSpeed=20
AckFrequency=1750
AckMinTime=4
AckDelay=1000
AckLevel=50
# Timeout=180
TimeoutLevel=80
CTCSSFrequency=88.5
CTCSSThreshold=30
# CTCSSHighThreshold=30
# CTCSSLowThreshold=20
CTCSSLevel=20
KerchunkTime=0
HangTime=7
AccessMode=1
# LinkMode=1 to remove almost all of the logic control
LinkMode=0
COSInvert=0
NoiseSquelch=0
SquelchThreshold=30
# SquelchHighThreshold=30
# SquelchLowThreshold=20
RFAudioBoost=1
MaxDevLevel=90
ExtAudioBoost=1
ModeHang=10

[POCSAG]
Enable=0
Frequency=439987500

[FM Network]
Enable=0
Protocol=USRP
LocalAddress=127.0.0.1
LocalPort=22346
GatewayAddress=127.0.0.1
GatewayPort=22345
PreEmphasis=1
DeEmphasis=1
TXAudioGain=1.0
RXAudioGain=1.0
ModeHang=20
Debug=1
LinkMode=1

[General]
Callsign=BI7JTA
Id=4600724
Timeout=240
Duplex=0
RFModeHang=300
NetModeHang=300
Display=Nextion
Daemon=1

[Info]
RXFrequency=439755000
TXFrequency=439755000
Power=1
Latitude=22.5676
Longitude=113.9873
Height=0
Location="SHENZHEN"
Description="China,Mainland"
URL=https://www.qrz.com/db/BI7JTA

[Log]
DisplayLevel=0
FileLevel=1
FilePath=/var/log/pi-star
FileRoot=MMDVM

[CW Id]
Enable=0
Time=10

[DMR Id Lookup]
File=/usr/local/etc/DMRIds.dat
Time=24

[NXDN Id Lookup]
File=/usr/local/etc/NXDN.csv
Time=24

[Modem]
Port=/dev/ttyAMA0
TXInvert=1
RXInvert=0
PTTInvert=0
TXDelay=100
RXOffset=0
TXOffset=0
DMRDelay=0
RXLevel=50
TXLevel=50
RXDCOffset=0
TXDCOffset=0
RFLevel=100
CWIdTXLevel=50
D-StarTXLevel=50
DMRTXLevel=50
YSFTXLevel=50
P25TXLevel=50
NXDNTXLevel=50
RSSIMappingFile=/usr/local/etc/RSSI.dat
Trace=0
Debug=0
Protocol=uart
UARTPort=/dev/ttyAMA0
UARTSpeed=460800
M17TXLevel=50
POCSAGTXLevel=50
FMTXLevel=100
AX25TXLevel=50
UseCOSAsLockout=0
```

bi7jta, 2024-04-15:
https://github.com/g4klx/MMDVMHost/blob/0e8788bdbec3314695fe367b4d9a85c7ba4b9738/Modem.cpp#L14

```
case MMDVM_GET_STATUS:
				// if (m_trace)
				//	CUtils::dump(1U, "GET_STATUS", m_buffer, m_length);

				switch (m_protocolVersion) {
				case 1U: {
						m_mode = m_buffer[m_offset + 1U];

						m_tx = (m_buffer[m_offset + 2U] & 0x01U) == 0x01U;
						bool adcOverflow = (m_buffer[m_offset + 2U] & 0x02U) == 0x02U;
						if (adcOverflow)
							LogError("MMDVM ADC levels have overflowed");
						bool rxOverflow = (m_buffer[m_offset + 2U] & 0x04U) == 0x04U;
						if (rxOverflow)
							LogError("MMDVM RX buffer has overflowed");
						bool txOverflow = (m_buffer[m_offset + 2U] & 0x08U) == 0x08U;
						if (txOverflow)
							LogError("MMDVM TX buffer has overflowed");
						m_lockout = (m_buffer[m_offset + 2U] & 0x10U) == 0x10U;
						bool dacOverflow = (m_buffer[m_offset + 2U] & 0x20U) == 0x20U;
						if (dacOverflow)
							LogError("MMDVM DAC levels have overflowed");
						m_cd = (m_buffer[m_offset + 2U] & 0x40U) == 0x40U;

						m_p25Space    = 0U;
						m_nxdnSpace   = 0U;
						m_m17Space    = 0U;
						m_pocsagSpace = 0U;
						m_fmSpace     = 0U;
						m_ax25Space   = 0U;

						m_dstarSpace = m_buffer[m_offset + 3U];
						m_dmrSpace1  = m_buffer[m_offset + 4U];
						m_dmrSpace2  = m_buffer[m_offset + 5U];
						m_ysfSpace   = m_buffer[m_offset + 6U];

						// The following depend on the version of the firmware
						if (m_length > (m_offset + 7U))
							m_p25Space    = m_buffer[m_offset + 7U];
						if (m_length > (m_offset + 8U))
							m_nxdnSpace   = m_buffer[m_offset + 8U];
						if (m_length > (m_offset + 9U))
							m_pocsagSpace = m_buffer[m_offset + 9U];
						if (m_length > (m_offset + 10U))
							m_m17Space    = m_buffer[m_offset + 10U];
					}
					break;

				case 2U: {
						m_mode = m_buffer[m_offset + 0U];

						m_tx = (m_buffer[m_offset + 1U] & 0x01U) == 0x01U;
						bool adcOverflow = (m_buffer[m_offset + 1U] & 0x02U) == 0x02U;
						if (adcOverflow)
							LogError("MMDVM ADC levels have overflowed");
						bool rxOverflow = (m_buffer[m_offset + 1U] & 0x04U) == 0x04U;
						if (rxOverflow)
							LogError("MMDVM RX buffer has overflowed");
						bool txOverflow = (m_buffer[m_offset + 1U] & 0x08U) == 0x08U;
						if (txOverflow)
							LogError("MMDVM TX buffer has overflowed");
						m_lockout = (m_buffer[m_offset + 1U] & 0x10U) == 0x10U;
						bool dacOverflow = (m_buffer[m_offset + 1U] & 0x20U) == 0x20U;
						if (dacOverflow)
							LogError("MMDVM DAC levels have overflowed");
						m_cd = (m_buffer[m_offset + 1U] & 0x40U) == 0x40U;

						m_dstarSpace  = m_buffer[m_offset + 3U];
						m_dmrSpace1   = m_buffer[m_offset + 4U];
						m_dmrSpace2   = m_buffer[m_offset + 5U];
						m_ysfSpace    = m_buffer[m_offset + 6U];
						m_p25Space    = m_buffer[m_offset + 7U];
						m_nxdnSpace   = m_buffer[m_offset + 8U];
						m_m17Space    = m_buffer[m_offset + 9U];
						m_fmSpace     = m_buffer[m_offset + 10U];
						m_pocsagSpace = m_buffer[m_offset + 11U];
						m_ax25Space   = m_buffer[m_offset + 12U];
					}
					break;
```

m0llc, 2024-04-16:
following.. i have this issue.